g992/flashforge-ad5m-5mpro-research #4: Connect single board computer

khayrulinnikita, 2024-01-26:
Hello! As I understand it, the host does not have enough resources to run a clean klipper. Is it possible to connect a single board computer like orange pi or raspberry pi to the N32G455 microcontroller? And flash the microcontroller with a clean assembly of klipper.

g992, 2024-01-26:
Yees, but u need some soldering skills. The other day I started researching in search of an opportunity to connect to his system, I found what I needed. I'm going to upload photos to /photos now, you can explore them. I plan to make a board that will be mounted on racks above the main board. it will have a CM4 minicomputer and everything you need to connect to the current system, as well as a CAN output for the print head

g992, 2024-01-26:
> Hello! As I understand it, the host does not have enough resources to run a clean klipper. Is it possible to connect a single board computer like orange pi or raspberry pi to the N32G455 microcontroller? And flash the microcontroller with a clean assembly of klipper.

I Write some sentences about this upgrade https://github.com/g992/flashforge-ad5m-5mpro-research/blob/main/upgrades/external_computer.md

jlodew0, 2024-02-25:
@g992 did you only connect rx and tx or also the gnd from the usb to ttl adapter?
I have the same adapters laying around as yours and I think I'm going that route instread of using the uarts of the rpi4 directly. That way, I can pry open a usb-hub with the 2 convertors and run a usb cable out of the enclosure. That way, I can keep my rpi external and easy to swap cards etc.

g992, 2024-02-25:
> @g992 did you only connect rx and tx or also the gnd from the usb to ttl adapter? I have the same adapters laying around as yours and I think I'm going that route instread of using the uarts of the [...]

gnd connected too, yes
if u use stock program versions, don`t remember send symbol to head controller (https://github.com/g992/flashforge-ad5m-5mpro-research/issues/6#issuecomment-1942928934)

closed 2024-08-01 by g992