g992/flashforge-ad5m-5mpro-research #5: 40mm fan addition to 5M base model, is it 12V?

sobriant74, 2024-02-01:
There is a spot on the 5M for screwing on a 40mm fan, if you remove the back, where it would pull air through the HEPA filter in the pro. I'm printing out the enclosure FF released a few days ago for the 5M, and I would like to add a fan there to pull air out for ventilation/thermoregulation.
There is a 4pin plug marked PLA FAN next to the yellow LED 2 pin plug. Do you know if that plug is 12V?
A NOCTUA NF-A4x20 PWM fan would fit there easily, if that plug works.
Thanks and keep up the great work, I'm going to work on mounting LED strips inside my enclosure next.

sobriant74, 2024-02-01:
BTW,
https://wiki.flashforge.com/en/knowledge_corner/printer-mod-and-diy  site with the files for printing the enclosure. 
https://wiki.flashforge.com/resource/models/printer-mod-and-diy/adventurer_5m_enclosure/ad5m-enclosure_for_3d_printing.zip

sobriant74, 2024-02-12:
ok, the 4pin JST port for the fan is 12v, and I'm adding a 40mm fan to the rear, as well as putting a 120mm fan on the inside for air circulation, once its enclosed in my new enclosure. They'll both plug into a jst y splitter, and I am going to mount two small 12v toggle switches to turn them on and off, since the 5M has no way, that I can see, to regulate turning those fans on and off.
If someone knows a setting to control those fans, please let me know. Thanks.

xblax, 2024-02-12:
I think it can probably be controlled via Klipper commands, see https://www.klipper3d.org/G-Codes.html#set_fan_speed
I think "Chamber Fan", "External Fan" and "Internal Fan" are not connected on 5M non pro. You can check out the printer config for Klipper, that was uploaded in this repo.

sobriant74, 2024-02-13:
> I think it can probably be controlled via Klipper commands, see https://www.klipper3d.org/G-Codes.html#set_fan_speed I think "Chamber Fan", "External Fan" and "Internal Fan" are not connected on 5M non pro. You can check out the printer config for Klipper, that was uploaded in this repo.

I'll check it out, I know the one of the fan ports is available on the 5M, but not sure which one. Guess I wont know until I try running Klipper.

sebmichel73, 2024-05-20:
On my AD5M the 4 pin JST is 22v on 1 pin and 24v on the other. None of the pins are 12v

closed 2024-08-01 by g992