g992/flashforge-ad5m-5mpro-research #7: SIlent fan

ps1x, 2024-02-05:
I went other route, i made Y-splitter with fan JST connector which is inserted into motherboard light output, then one tail goes to light, one back to PSU fan, so fan only works when light is on. This is much cheaper and eco-friendly then buying new PSU.

g992, 2024-02-05:
Nice mod! I installed mw psu not only because it has auto shutdown, but also because I don't trust Chinese power supplies :) I'm used to working with mean well, I fully support other power supplies, and I will be glad if you write a .md file about yours upgrade

closed 2024-02-07 by g992

anti-vaxxer, 2025-01-27:
Hi @ps1x @g992 !,
instead of buying new PSU would it be a good idea to open the PSU and add a thermal switch to the fan wire? so the fan turns on only when thermal switch reaches certain temperature.
Thermal Switch example https://www.aliexpress.com/item/1005006103476719.html