g992/flashforge-ad5m-5mpro-research #8: Building and flashing klipper to N32G455

g992, 2024-02-07:
Does somebody build + flash .bin file to n32g455? I try to made it, but i bricked mcus. If someone succesfully update klipper firmware, please write something here or here: https://klipper.discourse.group/t/nations-mcu-support-flashing/13893

jlodew0, 2024-02-18:
It seems like the N32G455 support is added since klipper 12.0
--> https://github.com/Klipper3d/klipper/blob/v0.12.0/src/stm32/Kconfig

config MACH_N32G455
        bool "Nation N32G455"
        select MACH_N32G45x

it's also in the release notes of klipper 12:
https://klipper.discourse.group/t/klipper-v0-12-0-release/11500

Several micro-controller improvements:
Support for new ar100 and hc32f460 architectures.
Support for stm32f7, stm32g0b0, stm32g07x, stm32g4, stm32h723, **n32g45x**, samc21, and

xblax, 2024-02-18:
@jlodew0 
Yes, because N32G455 is just an STM clone. They compiled MCU firmware for stm32f103xe and installed it on the N32G455. I can see this in Mainsail, because I think the MCUs send their version information via Klipper.

[image]

Klipper added official support only later, but it did work before. We can also see this in the Klipper Kconfig:
```
config MCU
    string
    ...
    default "stm32f103xe" if MACH_N32G45x
```

g992, 2024-02-19:
But if u just compile for stm32f103, u cant get access to analog periphery, and if u compile fw in new klipper version (0.12) mcu just cant boot

xblax, 2024-02-19:
@g992 
Did you manage to-rebuild the firmware with Klipper 0.11 and flash that or is anything self-built not bootable? Does Klipper need analog periphery for anything? And it works again after flashing the prebuilt MCU Firmware?

Maybe also check that clock frequency and reference are correct - but not sure if that would cause problems with booting.

g992, 2024-02-19:
The clipper needs analog peripherals at a minimum. for working with temperature sensors. I tried a lot, but nothing worked, when assembling under f103 the controller loads, but sometimes when I try to read data from the sensor it goes into error. on my printer I replaced the nation installed in the head with stm32f103, and just build a clipper for it, but I plan to replace the motherboard and head board to install your firmware, on which everything works on standard equipment

jlodew0, 2024-02-19:
This is some interesting reading: (Google translate ) https://blog.csdn.net/weiqifa0/article/details/126025874

Also: (but not specific to nations uc)

https://www.eevblog.com/forum/microcontrollers/which-of-the-many-stm32f103-clones-are-actually-usable/

mstrblueskys, 2024-02-25:
> The clipper needs analog peripherals at a minimum. for working with temperature sensors. I tried a lot, but nothing worked, when assembling under f103 the controller loads, but sometimes when I try [...]

I saw there was a release coming soon. Was there a breakthrough?!

g992, 2024-02-29:
I tried to flash it as is (I assembled a clipper for n32, flashed it using jlink) and nothing worked. I also tried to assemble it for s32f103, it worked, but the analog peripherals did not work. I didn't try again. as @xblax said, the system works fine with 0.11 clipper and there is no point in moving to 0.12