g992/flashforge-ad5m-5mpro-research #9: vOC sensor

AJolly, 2024-02-22:
I'm betting for number two in your photo section, it connects to a volatile organic compound sensor. It's listed in the klipper config

[temperature_sensor tvocValue]
sensor_type: MAX31865
sensor_pin: PD4
#spi_bus: spi4
#cs_pin: PD0
spi_speed: 500000
spi_software_sclk_pin: PE3
spi_software_mosi_pin: PE4
spi_software_miso_pin: PE5
min_temp: 0
max_temp: 2048
gcode_id: V

jlodew0, 2024-02-22:
nah, the name might suggest it. But since the sensor type is a max31865 is used for temperature, I will bet my head that it's just the internale chamber temperature and nothing more then that. laser based particle sensors (I have one for my arduino weather station) are to expensive for 3d printers.

g992, 2024-02-23:
this is not a temperature :)
if you disassemble the tvoc sensor on the pro version of the printer, you can find a really tvoc sensor, and... stm32f030, which receives tvoc values and simulates MAX31865, therefore the temperature displayed in the clipper from this sensor = the level of air pollution in the chamber
[image]

jlodew0, 2024-02-23:
Well at least it starts to get clear why there is such a price difference between the enclosed pro and non enclosed non pro version. I use a Nova SDS011 for pm 2.5 and pm10. I guess that is also why the parts cooling fan in the pro is on the opposite side to blow air over the sensor as well. (I guess) (the sds011 has a fan built in).
But what is flash forge using it for then? I know they have 2 filters, carbon and hepa. But if I'm not mistaken they are outtake fans? (what makes sense to filter the outgoing air). But do they also user the servo for recirculating the air until particles dropped to a certain level?  I'm a bit puzzled why they would add this. If the chamber get's to hot you have to vent, whatever the air. content is in the chamber. I wouldn't have expected this to be honest. also from a cost cutting perspective.

g992, 2024-02-23:
according to my observations, the printer turns on the exhaust fan at some level of pollution, there is no particle sensor, but just a tvoc, it is much cheaper than sensors that detect particles, but nevertheless it is enough to understand the level of pollution in the chamber
filters can be switched using a servo, they can work on air circulation in the chamber or just output to the outside, when printing with ABS plastic, it saves, since the printer is in the living room and thanks. filtration, 7-10 minutes after the end of printing, there is no odor in the chamber

jlodew0, 2024-02-23:
Allready printed these parts: https://wiki.flashforge.com/en/knowledge_corner/printer-mod-and-diy

but I allready regret not buying the pro version instead 😓 (for when I need something in abs)

closed 2024-08-01 by g992