g992/flashforge-ad5m-5mpro-research #13: Is there an AD5M LED installation tutorial?

EVSDELAT, 2024-03-05:
I've seen some friends install LEDs on their AD5M and want to ask how to install it

xblax, 2024-03-05:
There is a nice mod for it on Thingiverse: https://www.thingiverse.com/thing:6358773
I'm planning to also do this.

jacktruong, 2024-03-08:
I used some 120 LED/m 24V strips. I routed some 28awg wires through from the main board following the screen ribbon cable, along the top right side of the printer's frame, around the front, and to the top left side. Soldered the connections and insulated them. I need to clean up the cables a bit to prevent it from getting caught in the print head. I can post more pictures when that's done.

[image]

xblax, 2024-03-08:
The original cable routing can also be seen in a Flashforge maintenance video:
https://wiki.flashforge.com/en/adventurer-series/maintenance/ad5mpro-replace-the-led-light

Did anybody check what kind of connector is used on the Mainboard?

jacktruong, 2024-03-08:
> Did anybody check what kind of connector is used on the Mainboard?

I used some JST XH connectors I had and it worked