g992/flashforge-ad5m-5mpro-research #14: Backlight dimming/off

consp, 2024-03-06:
@xblax found the IOCTLs and I used them to create a [python package](https://pypi.org/project/ff-ad5m-backlight/0.1/) to make it more easy to deal with ([source](https://github.com/consp/flashforge_ad5m_backlight)).

When available discussion [here](https://github.com/xblax/flashforge_adm5_klipper_mod/issues/34)

Ross922, 2024-03-07:
Discussion link is broken - could you update? definitely interested in disabling the backlight... good thing its not an oled display or burn in would be instant lol

consp, 2024-03-07:
> Discussion link is broken - could you update? definitely interested in disabling the backlight... good thing its not an oled display or burn in would be instant lol

Discussion link is still private, when the klippermod becomes public (probably within a week) it will become available.

In the source is everything you need. If you have root just log onto the device, update the time and pip install ff-ad5m-backlight should be enough.

The original dtd allows for the ioctls to be used. More finegrained control can be achieved by disabling the display pwm control in the dtd and enabeling pwm7 but that is a bit too much for most users.