g992/flashforge-ad5m-5mpro-research #19: Mounting a RPI with screen on the printer

pelegw, 2024-03-21:
With the klipper mod now available, I wonder if anyone thinks of maybe adding a RPI with a small screen to the printer?
While klipper still runs on the main board, you can run klipperscreen on the RPI and offload this memory hog from the main board this way. 
You could also run multiple printers that way with a single screen through remote connection to moonraker. 

I'm working with a RPI+Screen I have leftover from another project, but my modeling skills kinda suck, so if anyone has an idea how to mount the assembly to the printer and can model it, I'll try out and show results.

mBlinkii, 2024-03-23:
I would replace the original display, either mount the RPI behind the new display or find a place for it at the back of the printer. You can model the whole thing with Fusion360, which is free for private use.

mBlinkii, 2024-03-23:
https://github.com/synthread/Klippventurer/issues/21#issuecomment-2016443956

xblax, 2024-03-24:
I would not recommend removing the original display. It provides the only external USB port and connects the Camera. Also I think if it's missing that could lead do various problems with the mod or Linux in general.

What you could do is to use the "lite" variant of the mod and mount another screen or tablet on top of the existing.

pelegw, 2024-03-24:
I'm trying out operating without a screen, so far two prints in, nothing bad is happening

FaqT0tum, 2025-04-28:
I'm preparing the project to replace all the electronics with improvements. This is the first part ready, these days I will also prepare the main board with SBC so that in the next few weeks I can create a guide. 
EBB36 + SKR Pico + RPi
[image]