g992/flashforge-ad5m-5mpro-research #22: QUESTION about the bed wiring

thhdragon, 2024-05-21:
hello me and another person on discord are attempting to do a board swap in the ad5m and since youve done a lot of research on it i figured you would be a great person to ask. have you by chance the unlabeled pins on the 30 pin flex cable that goes from the bottom of the bed to the mcu? trying to find which of the rest belong to the bed power/ground. for the bed it doesnt actually matter if you plug the + into the - on the mcu like how the hotend is im sure you already know. but getting the groups right IS important lol. by looking at it this is my guess. any help appreciated thanks. sorry if this is a weird place to post but i figured itd also make it so other people could see it
[image]

thhdragon, 2024-05-21:
little more background, so the only thing thats actually stopping us from throwing something like a pico in there is the weird bed ribbon cable. but the first 2 pins are thermistor, then the load sensor pins (tx, rx, gnd, vcc, det, rcs) then the rest are for the + and - for the bed heater. if we plug that flex cable into a $5 breakout board we can turn the thermistor cable and the bed load gauge wires into normal jst xh crimped connectors that can plug into a skr pico or something and then the rest into the screw terminals for the bed and boom you can plug it into any board you want. whether or not we can reuse the toolhead board is up in the air though but if we cant then we'd either need a different non weirdo proprietary toolhead board or since a pico has enough steppers its not a huge deal to run wires to the toolhead. theres no probe and we could do sensorless endstops. thats just 2 fans a heater/thermistor and an extruder wires to put through a cable chain.

NicoletoAerials, 2024-09-05:
Hi @thhdragon, I'm planning to swap the board on my printer and already have some electronics from another one. I’ve got a BTT Manta M8P mainboard and a BTT EBB42 e-board. However, I'm still trying to figure out the bed pinout. Did you end up doing the full swap? How did it go for you?

Astrias57, 2024-12-06:
Hello,
I'm looking for the reference of the bed connector. I'm not an expert in electronic but I'd like to try the same swap as @NicoletoAerials. Next step is to find how to communicate with the toolhead board.

Ziraya, 2025-07-26:
I spent a while looking really hard at the original image. I've made an assumption about the pin ordering on this socket, the machine is running so I can't chase it down right now. Everything in yellow is lower confidence, hard to see, too close, under silkscreens

Based on pours, pins 1 thru 11 appear to be ground, 12 through 22 main power coming from that mosfet with all the cooling vias.

pins 24 thru 30 are signals, if anybody recognizes these circuits maybe we can pin them down further

pin 23 goes into a via so I can't know what it is without removing the board and trying a lot harder, but notably pin 23 has a much thicker trace, 3~4 times the width, indicating that it is a power pin but much weaker than the pours, presumably a different voltage as well. 

Pins 28, 29, and potentially 30 seem to be part of a signal group. 30 goes into a via, it could go anywhere from there, but obvious candidates are R224 and D28. pins 28 & 29 are for sure being filtered by those 4 SMDs and interacting with D28. I think these are the best candidate for somebody who knows circuits like these to just know what they are immediately.

Pins 24 thru 27 go into that block of resistors, I'm not familiar with this arrangement, it looks like a group of 5 voltage dividers but with inputs/outputs on one end instead of the middle?

[image]

Ziraya, 2025-07-27:
I found a replacement board for the load cell, it uses an Aviaic HX711, [Sparkfun happens to have the datasheet](https://cdn.sparkfun.com/datasheets/Sensors/ForceFlex/hx711_english.pdf) and it outputs Serial. That other big IC could be wrapping it in something else, a lot of the devices around the AD5M seem to be SPI based on config files.

[image]

The load cell board runs an FFC to the PCB bed heater, which joins it into the main FFC, and there are labels there. I can make out:
main FCC, counting down from pin 30, odd pins
GNDA
GND
_O-1
_-TX

Load cell board counting down from high pin:
VCC
_-TX
O-1

Unfortunately getting at all these bits means [de-levelling the bed](https://www.youtube.com/watch?v=T4H2Mfyb9cA), and I have a lot of things I need to get done before I can do anything that disruptive. This part is available on some sites, but the images are not great

[image]

[image]

[image]

(this last one shows the traces much more clearly)

load cell board
1. GND
2. IO-1
3. IO-2
4. M3-TX? PCB-TX?
5. M3-RX? PCB-RX?
6. VCC

And these seem to map to the main FFC as
30 -> GNDA
29 -> NTC (thermistor)
28 -> GND
27 -> IO-2
26 -> IO-1
25 -> PCB-RX
24 -> PCB-TX
23 -> VCC