g992/flashforge-ad5m-5mpro-research #24: USB not working?

dangerousderek, 2024-07-30:
Hi! After looking at your detailed info I decided to try soldering on some connections to the two marked USB headers on the AD5M motherboard. Neither of them work and I am not sure why? I'm definitely not a pro, but I'm pretty sure they are connected properly. Any ideas why they wouldn't work in Klipper?
Thanks!!!

g992, 2024-08-01:
what modifications have you made exactly?

dangerousderek, 2024-08-01:
I just soldered headers on the two spots listed as usb on the main board. Then I tried wiring in a web cam to both spots and neither worked. The cam works when plugged into the port on the back of the display. I have two ad5m, one of which I got used and it didn't have a display. It works great as a headless now that I have done the klipper mod on it, but I really want a camera to see what's going on and with out the display there's no way to hook it up.