g992/flashforge-ad5m-5mpro-research #25: Connecting the 5MP main board to an external CPU?

NicoletoAerials, 2024-08-28:
Hi @g992,

Have you successfully connected the FF A5M board to an external CPU? I'm looking to connect it to a Raspberry Pi 4 running the latest versions of Klipper, Moonraker, and Mainsail. However, I have a couple of questions.

First, I'm uncertain whether this setup will allow me to connect both the 5M's e-board and the main board. Secondly, I'm considering using TTL to USB converters, but I wonder if it might be better to utilize the Raspberry Pi's GPIOs instead, as it has five available UARTs.

I would really appreciate your insights on this!

Thank you!

g992, 2024-08-29:
Hi, no, I've made several attempts, but without success, the best option is to install klipper (https://github.com/xblax/flashforge_ad5m_klipper_mod ) and use its web interface

NicoletoAerials, 2024-09-02:
Thanks for your response.

I'm currently using Xblax's Klipper mod, but I'm concerned about the 75% RAM utilization. I've encountered a few instances where the RAM was exhausted, especially during SHAPER_CALIBRATE, causing Klipper to reset.

To address this, I plan to replace the original mainboard and electronics with a BTT Manta M8P and an EBB 42 CANbus for the extruder. I believe these are the only boards I'll need to run a fully functional copy of Klipper on the A5MP and to future-proof the printer for software and firmware upgrades.

g992, 2024-09-03:
you can also use a native printhead board if you can assemble a klipper for the N32 processor on it (+ having some soldering skills, you need to remove 1 chip and solder another one). there is a UART output implemented, I was able to install a UART-CAN transceiver (scm3425) there as tests and it worked (RPi + CAN adapter saw the board, and the clipper was initialized)
But I couldn't flash the N32, so I put STM32F103 instead.

NicoletoAerials, 2024-09-05:
That’s a great idea! However, I’ve already purchased the EBB42 e-board and think I’ve found space to install it inside the printhead. I’m just waiting for it to arrive and will post some pictures soon. I’ll be using Xblax's Klipper configuration as a base, but I’ll adjust the input/output pins as needed. I still need to figure out the bed pinouts and build a breakout board to connect it to the Manta M8P.

NicoletoAerials, 2024-09-10:
> you can also use a native printhead board if you can assemble a klipper for the N32 processor on it (+ having some soldering skills, you need to remove 1 chip and solder another one). there is a [...]

@g992 I like your approach as it’s actually less invasive than mine. Is the STM32F103 a direct drop-in replacement? Do you have any pictures of your setup?

Thanks

mBlinkii, 2024-10-26:
i would also appreciate 1-2 pictures to see how you have implemented it @g992, it should be possible to swap the n32 1 to 1 with the stm32 as the pinout is the same. i would like to know how exactly you managed the communication with rpi.

i plan to swap the mainboard for an mp5, as there is not enough space in the print head for an ebb i wanted to use the original board and just swap the n32 for an stm32.

Jkbxhero, 2024-12-03:
I am also interested on where this went.  Mechanically this is such a robust machine for the price.  Firmware and software is awful and sometimes it just doesn't any to print.
I'm running 4 projects due by Christmas, research isn't something I have a lot of time for at the moment

Astrias57, 2024-12-07:
@g992 so, the eboard communicate using CAN with the CPU ? Sound interesting, BTT Manta M5P and M8P mainboard have CAN capabilities. I think I read on the github of xblax than the eboard need to be initialised using a small exchange through the serial port. Your solution looks better because it used the official klipper firmware. Is it hard to solder a STM32 in place of the N32 ?

g992, 2024-12-07:
board, unfortunately, connection via rs232, I installed f103, and flashed a regular uart there, it did not require any initializations, but I connected it to raspberry, I also connected the main microcontroller via uart, using the flashforge config. but I had problems with flashing the main processor, I could not assemble a stable firmware for n32, and in some places some functions did not work

Astrias57, 2024-12-08:
Thanks for yout fast answer. If I try to replace the N32 by a STM32F103, I imagine it's impossible to flash without a J-Link or ST-link connected to the SWD pins ont the toolhead board. It's impossible via RS232 ?
Have you describe the procedure you used somewhere ? If no, could you explain the main steps ? I need to solve 2 problems, the connection to the toolhead, and the connection to the bed and loadcells. I plan to by a bed and a toolhead board to try.

JacquesHumule, 2024-12-22:
Hello, can you give more information about why you can't flash the n32 ?

NicoletoAerials, 2025-01-26:
Klipper has reportedly introduced support for the Nation N32G455 chip. The configuration details for this chip on the FF5M e board should be as follows:

[image]

g992, 2025-01-27:
clipper has been supporting N455 for a long time, but I haven't been able to get it to work, meaning I've built .bin, flashed N32, but it didn't start (the klipper didn't see it). For experiments, I tried installing N32 on bluepill, but still couldn't get the klipper to start.

NicoletoAerials, 2025-01-29:
@g992 

What parameters did you use to build the STM32F103 klipper firmware for the native printer board? These are the ones I'm going to use. What do you think?

[image]

alund1990, 2025-03-02:
I've got mine running through a raspi 4 that connects to the moonraker instance on each printer. Use kiauh to install klipper, moonraker,  mainsail and whatever else you might want on there.  Edit the printer config file server and I believe host settings.  Add the ip address of printer(s). Then you should be able to go to the ip address of the raspi 4 control the printer(s) from there.

FaqT0tum, 2025-03-09:
I recently bought the A5M (not pro), the ram and obsolete klipper is tight for future expansions that I have in mind. If I can recover the eBoard and the original MCU (without SBC), using an external raspberry that communicates with 2 CP2102, I wonder! How do I recover the original display?

I could also decide to completely replace the original board with a SKR PICO (which I already have) and recover only the eBoard. Recovering the display would also be perfect! :)

ChaoticWeaponry, 2025-03-10:
Hi everyone! I'm having a lot of issues trying to get the e-board to connect to Klipper (running on a Pi4). I am able to communicate with the main MCU but not the e-board..

There weren't any resistors or pins to disconnect regarding the e-board? I'm guessing you're severing the lines by removing the resistors for the main MCU, but the guide doesn't mention anything other than soldering to the RS232 converter for the e-board.

Anything else I should try or things to look for? Appreciate the help..

g992, 2025-03-10:
Hello everyone Regarding the EBoard connection
Referring to the scan of the board (https://github.com/g992/flashforge-ad5m-5mpro-research/blob/main/photo/AD5M%20PRO%20Pcb%20Scan.png
I have provided a pinout of the RS232 converter, you can try to connect to it (if I remember correctly, 1 channel is used, that is, TX (AD5M) -> RX (USB UART) and RX (AD5M) -> TX (USB UART)
[image]

ChaoticWeaponry, 2025-03-10:
@g992 I’ll try the different pins when I get off work today. Thank you for the help!

hazama6, 2025-03-30:
> clipper has been supporting N455 for a long time, but I haven't been able to get it to work, meaning I've built .bin, flashed N32, but it didn't start (the klipper didn't see it). For experiments, I tried installing N32 on bluepill, but still couldn't get the klipper to start.

Wonder if ff is doing something weird with the eboard build since based on what they have for their "fork" they build a hex file instead of a bin

ChaoticWeaponry, 2025-03-30:
> > clipper has been supporting N455 for a long time, but I haven't been able to get it to work, meaning I've built .bin, flashed N32, but it didn't start (the klipper didn't see it). For experiments, I tried installing N32 on bluepill, but still couldn't get the klipper to start.
> 
> Wonder if ff is doing something weird with the eboard build since based on what they have for their "fork" they build a hex file instead of a bin. Also wondering is the STM32F103 a 1to1 swap?

Based on the pinouts the STM32F103 is a 1 to 1, but it's running at 72MHz when the N32 is 144MHz.

I'm a complete noob as far as coding goes, but maybe converting the klipper.bin file to hex (is that a thing?) would work?

hazama6, 2025-03-30:
> Based on the pinouts the STM32F103 is a 1 to 1, but it's running at 72MHz when the N32 is 144MHz.
>
> I'm a complete noob as far as coding goes, but maybe converting the klipper.bin file to hex (is [...]

Pretty much just different formats. What concerns me is toolchain differences since klipper generates code based on config files, different versions could produce different code for the same config. g992 seemed to have some luck with an older version of klipper but issues with adc on a newer version. I just wouldn't be suprised if there were other things ff did with their build that they refuse to share

ChaoticWeaponry, 2025-04-15:
Has anyone managed to successfully flash and boot  the E-Board with Klipper firmware?

I noticed in the datasheet that BOOT0 and BOOT1 are input pull-down.

The BOOT0 pin looks to be attached to a 10K resistor that goes to ground (through a via near the SWD port).

Maybe this has something to do with it not booting Klipper firmware?

hazama6, 2025-04-18:
> Has anyone managed to successfully flash and boot the E-Board with Klipper firmware?
>
> I noticed in the datasheet that BOOT0 and BOOT1 are input pull-down.
>
> The BOOT0 pin looks to be attached to a [...]

Tried today. Hooked it up to a pi running klipper via uart but couldnt see it

ChaoticWeaponry, 2025-04-18:
> Tried today. Hooked it up to a pi running klipper via uart but couldnt see it

How did you reflash the N32? Through SWD? Assuming with the klipper generated .bin?

hazama6, 2025-04-18:
> How did you reflash the N32? Through SWD? Assuming with the klipper generated .bin?

I havent. Not yet at least. Just powered the eboard and plugged into rx and tx while crossing my fingers hoping i didnt fry anything. The n32 already has firmware on it even if its not the right one. So it should show up assuming things are hooked up correctly

ChaoticWeaponry, 2025-04-18:
> I havent. Not yet at least. Just powered the eboard and plugged into rx and tx while crossing my fingers hoping i didnt fry anything. The n32 already has firmware on it even if its not the right [...]

Oh okay. With the stock MCU firmware you need to 'send' an acknowledge command for the e-board to initialize. How you're supposed to do that, I have no idea.

I remember seeing posts about it when development was happening for the klipper-mod.

ChaoticWeaponry, 2025-04-18:
> I havent. Not yet at least. Just powered the eboard and plugged into rx and tx while crossing my fingers hoping i didnt fry anything. The n32 already has firmware on it even if its not the right [...]

[https://github.com/g992/flashforge-ad5m-5mpro-research/issues/6#issuecomment-1942928934](url)

That's the development chat I was referring to. The boot process for the e-board MCU is really strange, I feel like we need to reverse engineer it in order to get it to boot klipper firmware.

hazama6, 2025-04-18:
> > > > > Has anyone managed to successfully flash and boot the E-Board with Klipper firmware?
> > > > > I noticed in the datasheet that BOOT0 and BOOT1 are input pull-down.
> > > > > The BOOT0 pin looks to be attached to a 10K resistor that goes to ground (through a via near the SWD port).
> > > > > Maybe this has something to do with it not booting Klipper firmware?
> > > > 
> > > > 
> > > > Tried today. Hooked it up to a pi running klipper via uart but couldnt see it
> > > 
> > > 
> > > How did you reflash the N32? Through SWD? Assuming with the klipper generated .bin?
> > 
> > 
> > I havent. Not yet at least. Just powered the eboard and plugged into rx and tx while crossing my fingers hoping i didnt fry anything. The n32 already has firmware on it even if its not the right one. So it should show up assuming things are hooked up correctly
> 
> [#6 (comment)](https://github.com/g992/flashforge-ad5m-5mpro-research/issues/6#issuecomment-1942928934)
> 
> That's the development chat I was referring to. The boot process for the e-board MCU is really strange, I feel like we need to reverse engineer it in order to get it to boot klipper firmware.

Thats super odd kinda wondering why. Like are they trying to get around some limitations or just obfuscating things? Just has a real weird smell

ChaoticWeaponry, 2025-04-18:
> Thats super odd kinda wondering why. Like are they trying to get around some limitations or just obfuscating things? Just has a real weird smell

If I had to guess, probably obfuscating things..

I've got a spare e-board that I'm tinkering with in my spare time.. At the moment I'm trying out wiring CAN to PA11/12 to a CAN transceiver. I need to buy a J-Link programmer then I'll mess around with hopefully wiping EVERYTHING off of the MCU and trying different klipper .bin files/bootloaders to see if anything 'sticks'.

ChaoticWeaponry, 2025-04-18:
> > > > > > > > Has anyone managed to successfully flash and boot the E-Board with Klipper firmware?
> > > > > > > > I noticed in the datasheet that BOOT0 and BOOT1 are input pull-down.
> > > > > > > > The BOOT0 pin looks to be attached to a 10K resistor that goes to ground (through a via near the SWD port).
> > > > > > > > Maybe this has something to do with it not booting Klipper firmware?
> > > > > > > 
> > > > > > > 
> > > > > > > Tried today. Hooked it up to a pi running klipper via uart but couldnt see it
> > > > > > 
> > > > > > 
> > > > > > How did you reflash the N32? Through SWD? Assuming with the klipper generated .bin?
> > > > > 
> > > > > 
> > > > > I havent. Not yet at least. Just powered the eboard and plugged into rx and tx while crossing my fingers hoping i didnt fry anything. The n32 already has firmware on it even if its not the right one. So it should show up assuming things are hooked up correctly
> > > > 
> > > > 
> > > > [#6 (comment)](https://github.com/g992/flashforge-ad5m-5mpro-research/issues/6#issuecomment-1942928934)
> > > > That's the development chat I was referring to. The boot process for the e-board MCU is really strange, I feel like we need to reverse engineer it in order to get it to boot klipper firmware.
> > > 
> > > 
> > > Thats super odd kinda wondering why. Like are they trying to get around some limitations or just obfuscating things? Just has a real weird smell
> > 
> > 
> > If I had to guess, probably obfuscating things..
> > I've got a spare e-board that I'm tinkering with in my spare time.. At the moment I'm trying out wiring CAN to PA11/12 to a CAN transceiver. I need to buy a J-Link programmer then I'll mess around with hopefully wiping EVERYTHING off of the MCU and trying different klipper .bin files/bootloaders to see if anything 'sticks'.
> 
> Curious to see what you find. I tried again but didn't get anything so i took my eboard out. Looks like communication is supposed to happen between 2 tpt3232e and i cant just plug uart in directly

If you’re trying to go directly to the e-board, I would think you could solder to the appropriate pads on the RS232 transceiver on the underside of the e-board. I have not tested that, I only soldered to the transceiver on the main board.

hazama6, 2025-04-18:
> If you’re trying to go directly to the e-board, I would think you could solder to the appropriate pads on the RS232 transceiver on the underside of the e-board. I have not tested that, I only [...]

Yeah I'd think that would work. Might just harvest the transceiver from the mainboard though

ChaoticWeaponry, 2025-04-21:
> So i went the transceiver route. Once i reply to the ready message im able to use the eboard. Only thing that breaks is the accelerometer. Now just gotta figure out the eboard firmware, assuming i can use make flash i figure i get one shot before i need to pick up an stm programmer

Good luck lol

I have a knockoff STM programmer and it couldn’t find anything when I wired up the SWD port, not sure if a legit one would work or not. I should have my J-Link by next weekend.

ChaoticWeaponry, 2025-04-21:
Hey guys!

I may have come across something that might help us out.

There’s a tool called Katapult that is basically a replacement bootloader, maybe we can combine this with Klipper firmware on the N32’s?

I’ll test this out when I get my J-Link programmer.

hazama6, 2025-04-25:
Hopefully works out better than it did for me. I tried pulling boot0 to vcc and flash katapult via uart but got nothing with stm32flash

FaqT0tum, 2025-04-28:
I'm preparing the project to replace all the electronics with improvements. This is the first part ready, these days I will also prepare the main board with SBC so that in the next few weeks I can create a totorial. EBB36 + SKR Pico + RPi[image]

ChaoticWeaponry, 2025-04-30:
I'm having a lot of issues using J-Link to erase/read/program the N32 MCU... I'm not sure if it's user error or perhaps there's some configuration register that prevents writing/reading from the chip?

At the moment none of the strategies J-Link uses to reset the CPU work, however I don't have the hardware reset wired in, only software.

I have BOOT0 pulled high to 3.3v and SWD is working/communicating fine.

I'm a bit in over my head at the moment, those with more knowledge with ARM or J-Link would be greatly appreciated.

ChaoticWeaponry, 2025-04-30:
> I'm having a lot of issues using J-Link to erase/read/program the N32 MCU... I'm not sure if it's user error or perhaps there's some configuration register that prevents writing/reading from the [...]

Update, so I'm able to reset the CPU via software ONLY with BOOT0 pulled low to ground. I'm still unable to erase the flash memory, returning with error -5 (failed to erase sectors).

hazama6, 2025-04-30:
Wonder if they have write protection

ChaoticWeaponry, 2025-04-30:
> Wonder if they have write protection

It sure seems like it. How would I go about removing it? I tried looking through the datasheet for the N32G45x but it’s a bit dense lol

hazama6, 2025-04-30:
Think you might be able to check if the protection bits are set... Maybe pulling boot0 up might work to clear it

ChaoticWeaponry, 2025-04-30:
> Think you might be able to check if the protection bits are set... Maybe pulling boot0 up might work to clear it

Same problem with BOOT0 at ground and 3.3v..

When I read the memory holding the option bytes it’s all FF.

Technically the entire memory dump is FF’s, so something is wrong or maybe the MCU is bricked.

hazama6, 2025-05-02:
Id say hook it back up to the printer and test it

FaqT0tum, 2025-05-05:
I created [this GIT](https://github.com/FaqT0tum/FlashForge_A5M_raspberry/) for the replacement of the complete electronics, you will also find some references to the various wirings perhaps useful even just for maintenance

all that's missing is the recovery of the original display (I don't know how much it's worth) and the load cells

ChaoticWeaponry, 2025-05-05:
> I created [this GIT](https://github.com/FaqT0tum/FlashForge_A5M_raspberry/) for the replacement of the complete electronics, you will also find some references to the various wirings perhaps useful [...]

This is awesome!

I used a BTT Touch 1.2 for my display and modified the enclosure top to fit it. No need for the stock display.

I believe the load cells in the stock Klipper-mod are used as temperature sensors? I’m not sure if that helps you.

hazama6, 2025-05-07:
Great work man. Was gonna swap my n32 with an stm32 but i may just go with this

FaqT0tum, 2025-05-07:
> This is awesome!
>
> I used a BTT Touch 1.2 for my display and modified the enclosure top to fit it. No need for the stock display.
>
> I believe the load cells in the stock Klipper-mod are used as [...]

yes exactly, probably the MCU (STC8G) of the load cell, emulates a MAX31856 but I'm not sure. unfortunately in the cfg there are pins that are not present in the N32G455 because it is a SOT64 and not 100. so I can't trace the original connection to understand how it works. my intent however remains to restore the load cells, I find it a convenient and functional function

FaqT0tum, 2025-05-07:
> Great work man. Was gonna swap my n32 with an stm32 but i may just go with this

I also thought about recovering the original Head board for formfactor but I don't like it at all. I had problems with electrostatic current, with failed prints. I solved it by isolating the board with aluminum foil

FaqT0tum, 2025-05-07:
small update, I managed to use the stock load cells, without doing much, but simply connecting (gnd, vcc, IO-1). everything is reported in the GIT. only the display remains out of the retrofit, no big deal

ChaoticWeaponry, 2025-05-07:
> small update, I managed to use the stock load cells, without doing much, but simply connecting (gnd, vcc, IO-1). everything is reported in the GIT. only the display remains out of the retrofit, no big deal

You’re awesome man!

Are they just acting as temperature sensors? Or how do they work?

FaqT0tum, 2025-05-08:
> > small update, I managed to use the stock load cells, without doing much, but simply connecting (gnd, vcc, IO-1). everything is reported in the GIT. only the display remains out of the retrofit, no big deal
> 
> You’re awesome man!
> 
> Are they just acting as temperature sensors? Or how do they work?

between the IO-1 pin and GND a voltage of +5v is created, when the strain gauges of the load cell detect a pressure on the plate, the voltage drops to a few mV. I simply thought that it could be used as a common digital endstop. It works great

ChaoticWeaponry, 2025-05-08:
> between the IO-1 pin and GND a voltage of +5v is created, when the strain gauges of the load cell detect a pressure on the plate, the voltage drops to a few mV. I simply thought that it could be [...]

I'll start buying all of the parts to make this soon (I'm considering a Manta so the SBC is integrated into the mainboard).

Question, is the voltage for the strain gauge analog? Or just a off or on switch?

Klipper-mod on the stock hardware the strain gauge appears to be analog, it registers different force with different 'temperatures'.

FaqT0tum, 2025-05-09:
> I'll start buying all of the parts to make this soon (I'm considering a Manta so the SBC is integrated into the mainboard).
>
> Question, is the voltage for the strain gauge analog? Or just a off or on [...]

It's analogue, it varies with the increase in pressure on the bed, but you can also use it as on/off as I did. It could also be used for other purposes to read an analogue value but for what was needed it was useless to integrate it

ChaoticWeaponry, 2025-05-09:
Good to know! I’ll start buying parts soon, if you would like help testing fitment of things or testing configs, let me know.

FaqT0tum, 2025-05-09:
> Good to know! I’ll start buying parts soon, if you would like help testing fitment of things or testing configs, let me know.

Of course thanks, if you have problems or you need a hand also write in the GIT section that I created, so as to create indirect help for other people as well. To date my A5M is fully functional, I have already made some prints. He doesn't seem to have problems. I only have to add a display and a webcam

KaruroChori, 2025-06-28:
I can try, but I don't have much confidence .